vtothsvk/AP-Home #8: Button pin died

vtothsvk, 2020-11-05:
**Describe the bug**
Button pin died. We don't know if it was caused by the board or the pin was already dead during the board testing.

**To Reproduce**
1. Flash main
2. Let the node boot
3. Measure Button pin output

4. Write the button pin HIGH in ` void setup( )`
5. Let the node boot
6. Measure Button pin output

**Expected behavior**
If the HW schematic is not at fault during the 1st measurement the pin will be LOW and during the second it will be HIGH.

xosuskyf, 2020-11-07:
ked dam git checkout buttonTest a potom git pull tak to vypise toto. (dal som predtym stash na ten kod co som mal akoze ja)

There is no tracking information for the current branch.
Please specify which branch you want to merge with.

pri ostatnych branchoch mi to v pohode stiahne danu verziu, prikaz git pull, pri tomto to vypise to vyssie.

vtothsvk, 2020-11-07:
pravdepodobne nie si v spravnom adresari

vtothsvk, 2020-11-07:
skus tieto veci skontrolovat ak nepomoze tak som o par min doma a kuknem na to

vtothsvk, 2020-11-07:
este ked tak skus git fetch a pak git pull

xosuskyf, 2020-11-07:
pri kompilacii buttonTest

AP-Home:98:9: error: 'button' was not declared in this scope

     if((button.clicks == 1)){

         ^

exit status 1
'button' was not declared in this scope

taktiez na riadku 63 button.Update();

oboje som vykomentoval a uz to ide skompilovat

xosuskyf, 2020-11-07:
v oboch pripadoch napatie 3.28 V, je to nova doska, kde button nebol pripojeny

vtothsvk, 2020-11-07:
> v oboch pripadoch napatie 3.28 V, je to nova doska, kde button nebol pripojeny

dobre, vyskúšal si ten pin mimo dosky? či keď má byť low tak je low a keď high tak high? (vieš, že či ten GPIO vieme ovládať vôbec aj keď sme to skúšali, ale nikdy nevieš)

xosuskyf, 2020-11-07:
ano skusal som to mimo dosky, ale mozno tym, ze to ESP bolo pripojene uz predtym na PCB dosku, kedze som testoval senzory, tak nie som si isty, ci uz ten pin nebol predtym odpaleny. Asi treba skusit naflashovat buttonTest na novu dosku, ktora nikdy nebola pripojena na PCB aby sme otestovali moznost ovladania pinu

vtothsvk, 2020-11-07:
Suhlasim.

EERCrso, 2020-11-09:
toto je aky button? Day/night alebo mute button ?

vtothsvk, 2020-11-09:
Nie, D/N je prepínač

xosuskyf, 2020-11-09:
kedze cely cas mame problem s tlacidlom vedla day/night prepianaca (takze ten mute button, v kode nazvany utility button, v D.T2.3 sprave nazvany programmable event button), tak to tlacidlo vedla testujem. Pin na tento button sme pravdepodobne odpalili pri prvej doske. Jedna sa o RX pin (D9, GPIO3). Aj v tasku je napisane, ze nevieme, co sposobilo, ze ten pin nie je odpaleny a ja nemam vedomost ze sme nieco riesli s day/night prepinacom. Ak mam riesit day/night prepinac, napiste.

Na uplne novej doske som nahral buttonTest kod a v oboch pripadoch testovania mi to dava hodnotu 3.28 V na RX pine (high). Musel som vykomentovat if z riadku 98 a buttonupdate z riadku 63. Prosim, @vtothsvk mozes skontrolvat, ci tym vykomentovanim som nieco nedodrbal a stale sa mal pin spravat pocas testu ako si uviedol v zadani?

Dakujem

vtothsvk, 2020-11-09:
S tym odkomentovanim si nemohol nič dodrbať, problém bude v samotnom core, keďže inicializuje RX pin na komunikáciu po sériovom porte, ktorý je na USB-čku. Čo je ale fest divné, že je pull-upnutý. Z mne neznámeho dôvodu je inicializovaný v open drain konfigurácii, čo mi pri UART-e v celku nedáva zmysel. Hrabem sa teraz v core a snažím sa to ohackovať. Ak sa mi niečo podarí dám update.

vtothsvk, 2020-11-09:
Dá sa to ohackovať, v prípade, že vyhodím celú inicializáciu Serial portu, čo znamená, že sa v takom prípade s nami doska nebude vedieť rozprávať cez USB. Príde mi to teraz ale ako bullshit, nakoľko sme to skúšali na sucho a všetko fičalo...

No nič @mato24101 máme tam ešte dajaký voľný pin?

xosuskyf, 2020-11-12:
Otazka znie, ci teda ten pin je funkcny pri sucasnom dizajne PCB boardu a staci len ohackovat ten pin po finalizacii produktu?

vtothsvk, 2020-11-12:
jop, ale sa akosi @mato24101 nevyjadril 😄

mato24101, 2020-11-16:
Ospravedlňujeme sa že som vám neodpísal ale teda voľné piny máme D10/GPIO1, D8/GPIO15, D12/GPIO10, D11/GPIO9 viem že niektoré z nich nieje možné využívať (myslím že D11 a D12) takže ak vieme skontrolovať či napríklad D10 nemá rovnaký problém tak to vieme v podstate jednoducho zmeniť

vtothsvk, 2020-11-16:
D8/GPIO15 by sme hypoteticky vedeli použiť, idem to testnúť.

vtothsvk, 2020-11-16:
Na D8 to nemá žiaden problém? @mato24101 neplánovali sme to aj pôvodne dať tam 😅 ?

EERCrso, 2020-11-25:
čo znamena bump ?

vtothsvk, 2020-11-25:
> čo znamena bump ?

bump je dummy správa/comment ktorá sa používa na pripomenutie konverzácie, v podstate taký poke, aby na to zapojení ľudia nezabudli

mato24101, 2020-11-26:
Takže odrezať stávajú u cestu a káblik na D8 hej?

vtothsvk, 2020-11-26:
> Takže odrezať stávajú u cestu a káblik na D8 hej?

áno

vtothsvk, 2020-12-06:
@mato24101 už hackol.

closed 2020-12-06 by vtothsvk