vtothsvk/AP-Home #11: No temperature, humidity, accelerometer and giroscope oputput

EERCrso, 2020-11-09:
**Describe the bug**
I cannot see information in  serial monitor about temperature, humidity, accelerometer and giroscope

**Logs and console output**
18:55:36.762 -> Motion:    0
18:55:36.762 -> Noise:     61
18:55:36.762 -> Smoke:     9
18:55:36.796 -> Gas:       48
18:55:36.796 -> Light:     41
18:55:36.796 -> Pressure:  0
18:55:36.796 -> Button:    0
18:55:36.796 -> STATUS: 00000001

**Evironment:**
 - OS: [e.g. Windows 10]
 - IDE used and its version (e.g. VSCode 1.48.2 with Arduino IDE 1.8.9)
 - Compiler toolchain and version (e.g. GCC - xtensa-esp32 v1.22.0)
 - Hash of the commit used or at least the branch/tag (e.g. [01d7c40](https://github.com/vtothsvk/AP-Home/commit/01d7c40c3d246cf4b00741955a6940bca525fcdd))

vtothsvk, 2020-11-09:
Added. @EERCrso or @xosuskyf pls test the [bme](https://github.com/vtothsvk/AP-Home/tree/bme) branch. Then ill merge it to main.

vtothsvk, 2020-11-12:
bol som nútený mergenút bme branch do mainu. @EERCrso alebo @xosuskyf  pls checknite ci to funguje.

EERCrso, 2020-11-16:
dnes sme  testovali kod a ide

closed 2020-11-16 by vtothsvk

xosuskyf, 2020-11-16:
Kod sice ide, ale vypisovat sa mi nepodarilo BME a navyse, neviem ako to mohlo ist, lebo sa mi marilo ze adresa je 0x77 (119)
Tak som si to skusil cez esp easy, kde mi urcite ficalo to BME a potvrdil som si adresu 0x77, pricom som to uz testoval priamo na PCB boarde, takze je to dobre aj pospajkovane. Prehodil som si adresu aj v kode, ale stale mi to nikde nevypisalo teplotu a ked som natvrdo si dopisal "getLastTemperature()" tak mi to vypisovalo 0. Ako sa vam podarilo rozbehat to BME chalani na nasom kode?
[image]