vtothsvk/AP-Home #17: Too big opening for debug and power supply

EERCrso, 2020-11-09:
**Describe the bug**
I would suggest to make the opening to fit more the USB port due to
1) light  shining out of the box
2) protection of main board
3) overall  profesional look 

[image]

xmirob, 2020-11-09:
The holes will be optimised in version 4. I need to have the device with the PCB inside or I need to have photo from the front orientation.

xosuskyf, 2020-11-12:
I suggest one thing, currently not to optimize the size of the hole to the ESP8266 microcontroller. Everybody form us has different cable with different ending  and therefore future debugging can be difficult. My cable is barely stuck inside of that hole (see the picture).
[image]

xmirob, 2020-11-26:
ok