vtothsvk/AP-Home #18: PIR senzor is not in stable possition

EERCrso, 2020-11-09:
**Describe the bug**
whet applying pressure on the outside part of PIR sensor the sensor is pushed back into the box

xmirob, 2020-11-09:
Hole will be optimized in version 4 but also glue in this case will be required to have a good stability. PIR sensor should be mounted on the second version of PCB.

EERCrso, 2020-11-10:
čo je PCB ?

xmirob, 2020-11-11:
PCB = Printed Circuit Board. Diameter of the hole was reduced in the V4 model. Glue will be necessary to apply to fix the PIR sensor properly. In the final version, the PIR sensor must be mounted on the base PCB board.

xosuskyf, 2020-11-12:
PIR will not be mounted on PCB because the mounting point is in total different corner of the casing. This issue has to be solved together with collimator issue https://github.com/vtothsvk/AP-Home/issues/44.

EERCrso, 2021-02-22:
fixed

closed 2021-02-22 by EERCrso