vtothsvk/AP-Home #19: Gas sensors in not stable possition

EERCrso, 2020-11-09:
**Describe the bug**
whet applying pressure on the outside part of GAS sensors the sensors are pushed back into the box, and due to small opening in the case they can but bushed back only with extra force, witch can lead to breaking of the box

xmirob, 2020-11-09:
We should meet and test this issue together, because, in my case it is stable.

EERCrso, 2020-11-10:
ked ich zatlačim dovnutra tak sa vcucnu.. ked to spadne na zem z boku tak sa zatlačia .. skus

xmirob, 2020-11-10:
Musia sa zatlačiť na doraz, potom, sedia správne a držia pevne. Ak boli nedostatočne zasunuté ako na fotografií, potom je to chyba nedostatočného zasunutia. Ide to však ťažko a hrozí zlomenie stredového predelu medzi otvormi, do modelu V4 som preto spravil o 0,1mm väčšie otvory.

xosuskyf, 2020-11-12:
It depends on material. In the case of PLA string material, I needed to enlarge little bit both holes on gas and smoke sensor  just to be able to fit them into the holes. In this material, the sensors are very stable.

EERCrso, 2020-11-25:
RIESENIE HOT GLUE

closed 2020-11-25 by EERCrso