vtothsvk/AP-Home #20: Noise sensor in not stable position

EERCrso, 2020-11-09:
**Describe the bug**
whet applying pressure on the outside part of noise sensor the sensor is pushed back into the box

xmirob, 2020-11-09:
The diameter of the hole will be optimized in version 4. Glue need to be added to ensure good stability. In version 2 of the hardware, all sensors where it is possible need to be on board. to ensure stability and eliminate cables.

xmirob, 2020-11-11:
Diameter of the hole was reduced in the V4 model. Glue will be necessary to apply to fix the noise sensor properly. In the final version, the noise sensor must be mounted on the base PCB board.

xosuskyf, 2020-11-12:
@xmirob in the final version it will not be possible to mount noise sensor directly. The reason is that the noise sensor and mounter header does not have enough space within the casing. I suggest that, the noise sensor will be directly soldered to the PCB board like in the case of first prototype.

EERCrso, 2020-11-25:
RIESENIE HOT GLUE

EERCrso, 2021-02-22:
nepoživa sa nakoniec

closed 2021-02-22 by EERCrso