vtothsvk/AP-Home #22: Humidity sensor moving freely

EERCrso, 2020-11-09:
**Describe the bug**
humidity sensor has no attachment part to the wall of the box. 

[image]

xmirob, 2020-11-09:
Fixation may be added in the design, but it needs to be fixed by the glue. In version 2 of the PCB, this need to be mounted on the PCB board. I also recommend to use another type of BME 280 PCB type.

EERCrso, 2021-02-22:
po  testoch  ekdy škatulu nie je možne otvoriť neveim posudit či to bolo fixnute. tak či tak humidity senzor je nepoužitelny, taktiež je nepoužitelne meranie teploty ked je čislo uzavrete v škatulke takymto sposobom.. vnutorna teplota škatule je cca 34 stupnov zatial čo okoli ma cca 21 .. zotvačnost s akou sa tepota meni je uzavretim senzorov velmi pomala radovo ak nejaka vobec....