vtothsvk/AP-Home #23: No day night switch

EERCrso, 2020-11-09:
**Describe the bug**
There is no day night switch.

mato24101, 2020-11-11:
Day night switch je vo fáze že 10ks máme u nás vkancelárií a bolo by prípadne dobré overiť či sa dokáže vmestiť do krabičky spolu s ostatnými vecami ktoré sa tam už nachádzajú. Prípadne ak sa nezmestia bol oby dobré vybrať iný dizajn prepínaču a následne možno upraviť aj krabičku (najme sa bavíme o uchytení a otvore pre prepínač). Zatial máme nakúpené tieto prepínače https://www.gme.sk/packovy-spinac-smts-102

EERCrso, 2020-11-11:
rovno to dajme teda mirovi ne ?

xosuskyf, 2020-11-25:
Day night switch was added to all boards. This issue should be solved.

EERCrso, 2020-11-25:
este nie je naprogramovany

xosuskyf, 2020-11-25:
tuto je to zmienene, nemam zariadenie na testovanie, inak by som to skusil a dal zelenu na pull request.
https://github.com/vtothsvk/AP-Home/issues/64

EERCrso, 2020-11-25:
jaj ok

closed 2020-11-25 by EERCrso