vtothsvk/AP-Home #28: Weak material of casing

EERCrso, 2020-11-09:
**Describe the bug**
 The material has a high tendency to break. It is quite fragile

[image]

[image]

[image]

[image]

xmirob, 2020-11-09:
It is designed in the case of wearables to press the part in the hole to open the cover safely. Anyway, hole click system will be changed to the system as it is in AP Nurse Home Base. Printed arrow will show where to open the cover. Holes for screws will be removed in version 4 completely. Only holder will be used for fixation to the wall. The thin layer on the wearables for diodes will be thicker and will be fixed by separators.

EERCrso, 2020-11-10:
cele by to chcelo spravit hrubšie už sa mi to cele rozlamalo skoro

xmirob, 2020-11-10:
Walls, bottom and top side are 0,2 mm thicker in V4 Base model. Screw holes in Base V4 model are removed.

xosuskyf, 2020-11-12:
The V3 version of casing is much harder with standard string material. Also in the case of string material, the screw holes in the sensor board casing are better for stabilisation of the PCB board in the casing.

EERCrso, 2020-11-16:
hey to SLA je krehke jak fras 
[image]

EERCrso, 2021-02-22:
problem vyriešeni odsttranenim SLA tlače

closed 2021-02-22 by EERCrso