vtothsvk/AP-Home #30: Motion alert problem prototype1

EERCrso, 2020-11-09:
**Describe the bug**
prototype p1 case V3 from FIlip that I aquired 28.11.2020 has following problem

Device has permanent motion alert even if I turn off measurement of motion (PIR sensor ) I still get motion alert
What have you tried to diagnose or workaround this issue?
serial monitor

**To Reproduce**
Steps to reproduce the behavior e.g.:
1. turn on sensor board device
2) read serial monitor

**Expected behavior**

no to have motion 0 when uncommenting

//uncommenting the next line disables PIR sensor update
#define _DISABLE_PIR

**Logs and console output**

18:55:36.762 -> Motion:    1  
18:55:36.762 -> Noise:     61
18:55:36.762 -> Smoke:     9
18:55:36.796 -> Gas:       48
18:55:36.796 -> Light:     41
18:55:36.796 -> Pressure:  0
18:55:36.796 -> Button:    0
18:55:36.796 -> STATUS: 00000001    // This code indicates motion alert

all thresholds on other sensors are  at maximum to not mess up 

**Evironment:**
 - OS: [e.g. Windows 10]
 - IDE used and its version (e.g. VSCode 1.48.2 with Arduino IDE 1.8.9)
 - Compiler toolchain and version (e.g. GCC - xtensa-esp32 v1.22.0)

I have also tried commenting 

bool AP_Nurse::getLastMotion(){
    //return (this -> ap_node.lastMotion);    / THIS LINE
}

how ever I get motion 0 but motion alert is still on

vtothsvk, 2020-11-09:
Temporarily fixed by [dcf0144 merge](https://github.com/vtothsvk/AP-Home/pull/9).

vtothsvk, 2020-11-09:
No to je divné, nakoľko by to ten pin vôbec nemal čítať.

vtothsvk, 2020-11-09:
@EERCrso  našiel som zradu, kompiler ma nepochopil 🤣 boha to gcc... Mea culpa, fixnem asap. Pridám to potom pre všetky merania.

EERCrso, 2020-11-09:
Ok v pohode super dik

vtothsvk, 2020-11-09:
Malo by byť fixnuté poslednym mergeom [589f1b0](https://github.com/vtothsvk/AP-Home/pull/33).
Pozor direktíva je v novom súbore **Sensor_enable.h**. Neskôr tam pridám direktívy aj pre ostatné senzory.

EERCrso, 2020-11-09:
Zajtra rano to pretestujem a dam vedieť.. ak to fachči možme isť parcialne na Test cases... diki

closed 2020-11-10 by vtothsvk