vtothsvk/AP-Home #31: PIR sensor saturation.

vtothsvk, 2020-11-09:
**Describe the bug**
PIR sensor gets saturated in daylight.

**Additional context**
Do we have a solution? @mato24101 can we make smth about it from HW point of view? @xosuskyf other PIR sensor (eg. the M5Stack ones) or some other solution?

xosuskyf, 2020-11-25:
PIR dedicated for waspomte technology (Paralax 555-28027) is working like a charm, however the casing does not have proper opening for this sensor and also this sensor is huge 2.5x3.5x2.5 cm.
[image]

vtothsvk, 2020-11-25:
@xosuskyf možno blbosť, ale skúsil sa už aj iný kus z tých PIR-ov, čo sa pôvodne na Home používali? 😅

xosuskyf, 2020-11-25:
Skusil som 3 rozne

EERCrso, 2020-12-02:
PIR ktory dodal filip najnovsi je  v poriadku  Uzatvaram

closed 2020-12-02 by EERCrso