vtothsvk/AP-Home #42: Problem s parovanim bez alertu na bracellet board

EERCrso, 2020-11-10:
**Describe the bug**
Hodinky sa nesparuju pokial nie je alert

**To Reproduce**
Steps to reproduce the behavior e.g.:
zapni bracelet board
bracelet bord sa zapne a nevydáva alert
zapni hodinky
hodinky sa nesparuju nikdy
vyvolaj alert
hodinky sa az nasledne sparuju

**Expected behavior**
sparovanie vzdy aj bracelet board nevydava alert

vtothsvk, 2020-11-10:
Jop, ono je to tým, že encoder spí (neposiela správy) kým nedôjde k zmene statusu. Čo znamená, že prvú správu vyšle až keď nastane prvý alert.

Possible fix:
1. Párovacie okno pri zapnutí boardu
2. Periodické vyslanie správy

Spravím kód pre oba prípady a uvidíme, ktorý vyjde lepšie.

xosuskyf, 2020-11-10:
Pod prvym krokom "zapni bracelet board" myslis asi sensor board, lebo nikde v krokoch nespustame senzoricku dosku.
To, ze sa to teraz nesparuje je asi kvoli tomu, ze TE pin teraz nevysiela stale a iba pri zmenach. Overime.

vtothsvk, 2020-11-10:
Nahodil som tam nakoniec aj aj. Tj. keď sa device zapne je tam 5s okno, kedy posiela správy jak divý nech sa spáruje a pak sa ešte každých 5s ozve, keby náhodou spojenie vypadlo/v prípade resetu hodiniek netrebalo fiurt aj krabicu resetovať.

Pridal som to posledným mergeom do [main](https://github.com/vtothsvk/AP-Home)u ([#45](https://github.com/vtothsvk/AP-Home/pull/45)).

EERCrso, 2020-11-10:
ano tak som to myslel.... ako si pisal FIlip..... otestovane. FUnguje to len v pripade ak je zariadenie pripojene k PC a sledujem monitor v opačnom pripade nie..  deje sa nasledovne:
sensor board odpojeny od PC
- zapnem bracellet board
- zapnej hodinky.. hodinky su nesparovane svietia dve diodi a vrčia jak dive
- po chvili zasvieti len jedna dioda a vrčia menej

vtothsvk, 2020-11-10:
Bol inicializovaný len jeden pin Encoderu, fixol som to posledným pushom, je to na maine. Čím môže byť, že to funguje len keď to máš pripojené si nie som istý. Skús teraz uvidíš čo to spraví.

xosuskyf, 2020-11-12:
hm, nema @EERCrso porbelm s tym, ze maju to iba pripojene na kabli, bez hlavneho napajanie, ako sme to mali predtym, ked tie ostcilacne frekvencie, kvoli znizeniu napatia na baterkach nesedeli? Ake mate @EERCrso napatie na oboch bateriach?

EERCrso, 2020-11-16:
Chybaju tam nejake subory nato aby to skusil 
[image]

EERCrso, 2020-11-16:
Filip ne tam problem neni

vtothsvk, 2020-11-16:
@EERCrso mea culpa, som vás zabudol upozorniť, že som aktualizoval setup prostredia.
Checknite [dokumentáciu](https://github.com/vtothsvk/AP-Home/blob/main/Documentation/Programming.md) konkrétne **Install dependecies**
Potrebujete si nainštalovať knižnice pre bme280
(@xosuskyf pls checkni tiež)

EERCrso, 2020-11-16:
Jo dakujem.. stale to nefunguje hodinky sa paruju ale miesto toho aby sa uhludnili zacne svietit len jedna dioda a menej vybrovat.. to iste ked odpojim debug a mam ich len na napajani

EERCrso, 2020-11-16:
neveim ci to nazvat vobec sparovanie..... tento problem je aj cez debug aj bez debug

vtothsvk, 2020-11-16:
Sry blbo som mergol jeden pull request. Posledným mainom by to malo ísť.

EERCrso, 2020-11-16:
Uz to ide, testovane teraz na mojom PC (MC).

closed 2020-11-16 by EERCrso