vtothsvk/AP-Home #63: Unable to Pair bracelet board and sensor board + strange behaviour

EERCrso, 2020-11-23:
**Describe the bug**
Hodinky nejde sparovat +  spravaju sa zvlastne popis v To reproduce

**To Reproduce**
Steps to reproduce the behavior e.g.:
Nahral som na bracelet board najnovsi firmware
spustim zariadenie
neparuje 
resetoval som senzor board mnohonasobne a nikdy sa nesparuju
Niekedy hned po resete prestane bracelet borad vybrovat a vtedy mam pocit že sa sparoval ale nie je to tak ked potom vytovrim alert napriklad pressure senzorom tak hodinky nereaguju a nevybruju

**Expected behavior**
ze sa sparuju

**Evironment:**
 - OS: [e.g. Windows 10]
 - IDE used and its version (e.g. VSCode 1.48.2 with Arduino IDE 1.8.9)
 - Compiler toolchain and version (e.g. GCC - xtensa-esp32 v1.22.0)
 - Hash of the commit used or at least the branch/tag (e.g. [01d7c40](https://github.com/vtothsvk/AP-Home/commit/01d7c40c3d246cf4b00741955a6940bca525fcdd))

EERCrso, 2020-11-23:
Je to divne  asi po hodine som vsetko zopakoval a sparovalo hodinky  v pripade že som nebol pripojeni k PC

xosuskyf, 2020-11-23:
Zmerajte napatie na baterkach. Mozno uz moc kleslo a zacina to blbnut. Toto mozno bude dobre aj zdokumentovat ako hranicnu podmienku parovania. Mne fungovali vsetky zariadenia v piatok, co sa tyka parovania s tymto braceletom.

EERCrso, 2020-11-23:
napätie je 5.8 V ... momentalne paruju ...

EERCrso, 2020-11-24:
dnes opat paruju zatvaram ked sa zopakuje otvorim

closed 2020-11-24 by EERCrso