vtothsvk/AP-Home #64: Zapracovanie treshold parametrov pre režim day night

EERCrso, 2020-11-24:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
potrebovali by sme do kodu miesto kde budeme nastavovat tresholdy pre day / night režim momentalne je tam len jedno všeobecne miesto

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

vtothsvk, 2020-11-24:
Testnite pls [dayNight](https://github.com/vtothsvk/AP-Home/tree/dayNight). Pozmenil som data update a alert handling tak, aby bral do úvahy D/N režim a pridal direktívy pre night mode thresholdy.

EERCrso, 2020-11-25:
Ide to ok  ale ak je to v rezime Night teda vypinač v polohe vzdialenejšej od diery na button tak hadze STATUS spravne ( mal davat noise alert) ale vypis je 

09:43:46.221 -> Noise:       0
09:43:46.221 -> Smoke:       0
09:43:46.221 -> Gas:         0
09:43:46.221 -> Light:       0
09:43:46.221 -> Pressure:    0
09:43:46.221 -> Temperature: 0.0
09:43:46.221 -> Humidity:    0.0
09:43:46.221 -> APressure:   0.0
09:43:46.221 -> Button:      0
09:43:46.221 -> STATUS: 00000010

ak vypinač prednem na day a potom spat na night tak vypis je uz ok..

vtothsvk, 2020-11-25:
@EERCrso v dennom režime noise alert funguje?

vtothsvk, 2020-11-25:
@EERCrso ak som teda správne pochopil, tak alert vypíše, ale nevypíše hodnotu?

xosuskyf, 2020-11-26:
Testoval som a zistil som nasledovne. @EERCrso  co pise je pravda, a doplnim s cim som sa stretol este. Je to pruser s danym switchom. Takze, ak zacnes so switchom zopnutym tak, ze dava output high na ESP, tak vypise 0. ESP vadi, ked po restarte/boote ma na tom pine high. Prehodis na output low na dany pin a zacne bezat, a ked prehodis naspat, tak niekedy to prepne direktivy do druheho modu. Co je pruser je, ze ono to resetne dosku velakrat pri switche z output high na -> output low (z output low na output high som nenasiel problem). Musi si clovek najst grif, aby to neresetlo. Raz sa mi stalo, ze po prepnuti som mal infinite loop of restarts na ESP. Neviem co to robi, ale predpokladam, ze vo vnutri switchu dojde na mikro sekundu k mini skratu, a preto to vsetko resetuje.

vtothsvk, 2020-11-26:
No ja som to testoval tak, že som prepájaním kábla simuloval prepínač, a nerobilo mi to zmienené problémy. 
Naschval som testoval, čo to robí s bootom v jednej aj druhej polohe a pod. , no pravda netestoval som to už s hotovým FW, čo by ale funkcionalitu toho switchu nemalo ovplyvniť. Je ale možné, že spínač pri prepínaní na chvíľu skratuje šťavu a zem, čo môže celkom pekne vadiť doske. 
@xosuskyf , @mato24101 vedeli by ste pls zemrať, či v istej polohe ten spínač neskratuje všetky jeho piny?

xosuskyf, 2020-11-26:
Zmerat asi nie, ale ako som vyssie napisal, raz som prepol prepinas a dokolecka to resetovalo dosku, takze vtedy predpokladam, ze bol prave ten skrat.

vtothsvk, 2020-11-26:
> Zmerat asi nie, ale ako som vyssie napisal, raz som prepol prepinas a dokolecka to resetovalo dosku, takze vtedy predpokladam, ze bol prave ten skrat.

viď pls [D/N switch possible short circuit](https://github.com/vtothsvk/AP-Home/issues/72)

vtothsvk, 2020-11-26:
A páni pls vyjadrite sa, ohľadom FW zapracovania, nech môžem uzavrieť [PR](https://github.com/vtothsvk/AP-Home/pull/66) a túto issue.

EERCrso, 2020-11-26:
FW prešiel testom v poriadku

closed 2020-11-26 by vtothsvk