vtothsvk/AP-Home #65: Krit na light senzor ASAP

EERCrso, 2020-11-24:
**Describe the bug**
momentalne je light senzor plne exponovany a v zavere bude určite prekrity nejakym polopriepustnym mliečnym plastom pre akymkolvek nastvovanim tresholdov je potrebne vediet akym, akej hrubky, mat zo  fyzicky,  kedze toto sposobi utlm dopadajuceho žiarenia na senzor a ine hodnoty tresholdov.

JB

xosuskyf, 2020-11-25:
V prvom kroku navrhujem vhodne zasunut tento senzor do dedikovaneho otvoru, aby bol zarovno so stenou casingu. Ak to nepojde ohnutim pristupovych kontaktov, tak aby ste nic neposkodili, tak privodove kontakty natvrdo skratte odcvaknutim (jeden kratky light senzor nas nezabije). Neviem ci stihame nieco prekryvat polopriepustnym plastom, kedze dalsiu spravu musime napisat v priebehu decembra.

EERCrso, 2020-11-25:
ide len o to že nebudeme mat relevantne hodnoty do nejakej spravy ak to nebudeme testovat tak ako to bude  v realite ...

EERCrso, 2020-12-02:
mame foliu  od filipa a snou budeme testovat... zatvaram

closed 2020-12-02 by EERCrso