vtothsvk/AP-Home #67: Saturovany noise senzor

EERCrso, 2020-11-25:
**Describe the bug**
Noise senzor  v tichu dava hodnoty okolo 235

What have you tried to diagnose or workaround this issue?
Nie nemam ako

**To Reproduce**
Steps to reproduce the behavior e.g.:
1.  nahrat posledny commit main https://github.com/vtothsvk/AP-Home/commit/9783c2de9a46e4b60074e002fa2119b8dc4565a0
2. zapnut zariadnie ( siva skatula co mam  na test )
3 sledovat monitor

**Expected behavior**
odost nižšie hodnoty ak je ticho

**Logs and console output**
08:43:36.746 -> Noise:       235
08:43:36.746 -> Smoke:       9
08:43:36.746 -> Gas:         38
08:43:36.746 -> Light:       12
08:43:36.779 -> Pressure:    0
08:43:36.779 -> Temperature: 26.0
08:43:36.779 -> Humidity:    28.7
08:43:36.779 -> APressure:   100344.9
08:43:36.779 -> Button:      0
08:43:36.779 -> 0

**Evironment:**
 - OS: [e.g. Windows 10]
 - IDE used and its version (e.g. VSCode 1.48.2 with Arduino IDE 1.8.9)
 - Compiler toolchain and version (xtensa-lx106-elf-gcc)
 -  https://github.com/vtothsvk/AP-Home/commit/9783c2de9a46e4b60074e002fa2119b8dc4565a0

xosuskyf, 2020-11-25:
Mate pustene hlavne 5 V napajanie? Ak ano svietia dve diody na noice senzore? Nastavte potenciometrom na noice senzore stav, kedy druha dioda zhasne alebo sa rozvieti, vtedy dosiahnete najvyssiu citlivost noice senzora, pripadne vyskusajte tocit potenciometrom pocas toho ako nacitavate hodnoty a mali by ste uvidiet rast alebo pokles hodnot priamo v loge realtime. Kazdopadne treshold sa bude odvijat od miesta si myslim, kedze reakcia na silny zvuk bola zmena hodnoty do 20 a menej, ked som to testoval.

EERCrso, 2020-11-25:
5V napajanie je pustene......... žiadne diody nesvietia .... ale potenciometrom ked točim tak sa to meni diki

EERCrso, 2020-12-02:
Funkcny test NOISE SENSOR:

- je mozne nastavit jeho hodnotu pomocou potenciometra
- je napajany cez 5V na sieti / na baterke + debug necg vidime co sa deje
- vidno zmenu hodnot pri manupilacii s potenciometrom priamo v monitore serioveho portu
- SENZOR NEREAGUJE NA ZMENY HLUKU (skusali sme priblizne rozdiel 20 dB)

vtothsvk, 2020-12-02:
@EERCrso skúste ho pls zmerať na sucho, aby sme vedeli kde je zrada. Ak bude bez pripojenia meniť výstup tak je zrada v kóde, ak nie tak v senzore.

EERCrso, 2020-12-03:
No po prepajkovani druheho senzoru reaguje.. bohužial jeho spravanie aj pri nastraveni najnižšej citlivosti je chabe.... reaguje alen na extremne hlikove zmeny v tesnej blizkosi resp vibracie priamo na senzor.. na realnu zvuk v okoli takmer nereaguje.. nebude sa dat použiť...

closed 2020-12-03 by EERCrso

xosuskyf, 2020-12-03:
Skusal som pri senzore kricat a obcas vyskocila hodnota tak o 7-10. Mozno nam to fakt nepojde, ale ked na nete robili s tym senzorom rozne hudobne equalizeri typu ako windows media, ze ti to na zaklade zvukov v hudbe robi rozne obrazce a podobne, tak by mozno s tym nieco slo spravit. Kazdopadne, ked ESP nezachyti takuto kratku zmenu, tak suhlasim, ze to bude nepouzitelne, alebo bude pouzitelne iba vtedy, ked nebudeme pouzivat analogovy vstup ale digitalny, kde bude manualne potenciometrom nastaveny treshold.