vtothsvk/AP-Home #69: Create internal logic for APnurse HOME

EERCrso, 2020-11-25:
**Describe the solution you'd like**
Create internal logic for APnurse HOME and two level alert system acording 
D.T2.2.3 Table
[image]

vtothsvk, 2020-11-29:
@EERCrso teraz som si všimol, nie je v tej tabulke chyba? AP4 night/movement by nemalo byť critical condition?

EERCrso, 2020-12-01:
Ahoj ano je tam chyb ma tam byt alert 2

EERCrso, 2020-12-01:
teda cerveny alert

vtothsvk, 2020-12-05:
Takže, dokončil som ten config. Nájdete to v [configTest](https://github.com/vtothsvk/AP-Home/tree/configTest). Ak to bude testnuté mergnem (https://github.com/vtothsvk/AP-Home/pull/83)

EERCrso, 2020-12-08:
len pišem z našej diskusie aby sa nestratila info... light senzor sa neda vypnut...   light senzor sa neda vypnut samostatne len s plynmy, ktore ale potrebujeme furt... vravel si že vieš spraviť že ak sa nahodi treshold 255 na light tak light nebude nikdy davať alert... momentalne dava

vtothsvk, 2020-12-08:
no je to tam nahodené, nemal by dávať alert

EERCrso, 2020-12-08:
nefunguje to ak na light nastavime  255  a dosiahneme hodnotu 255 system dava light alert

vtothsvk, 2020-12-08:
lebo je to v maine, config čaká na schválenie a mergne sa tiež do mainu

EERCrso, 2020-12-08:
ok ideme testnut

EERCrso, 2020-12-08:
ten light funguje  super

EERCrso, 2020-12-08:
Tu su zakladne poznamky z prvych testov uz upravenych verzii + configu:
- **senzory LDR a PRESSURE nevyvolaju nikdy alert level, vidno vsak ze funguju cez STATUS**
- upraveny event: AP1 - night - leave bed - Pressure Drop = Alert 2 (namiesto Alert 1)
- zruseny event: AP1 - night - fall from the bed - Movement + Pressure Drop = Alert 2 -> je to logicka duplicita a zbytocna komplikacia
- event: AP1 - night - walking in the room - Movement + Light = mal by sa rovnat Alert 1, namiesto toho chyti vzdy len event: night - waliking in the room - Movement = Alert 2 -> tu je otazka ako to vobe realizovat aby tam ten alert niekedy skocil, jedine asi nejaky delay ci bude v najblizsich sekundach aj vystup z Light, neviem 
- timer stuck in bathroom nefunguje, nie je mozne dostat sa do alertu lvl 2, verzia AP2/5 - DOOR - ma len alert 1
- AP4 - OK
- AP6 - vyzera ok, treba plyny lepsie testnut este
- AP7 - rovnake problemy ako AP1/3

su upravene aj podklady na gire - v dokumentacii novy excel