vtothsvk/AP-Home #71: Zmena baterie - Case -

EERCrso, 2020-11-25:
**Describe the solution you'd like**
miro bude sa uplne menit tvar velkost baterie kedže sučasna nevyhovuje požiadavka nabijania a sučasneho napajania senzor board. Treba s tym ratať pri dalšom navrhu Case. Ked bude funkčna a otestovana dame ti ju na implementovanie Taktiež treba ju implementovat tak aby kabel ktory z nej pojde do napajanie sentzor board nebolo možne  odpojit a teda bol schovany vo vnutry zariadenia. 

Odporučam aj celu bateriu schovat do vnutra Case a cele to uzavrie ( nech to nikoho nedraždi to rozpajať alebo ujebat baterku ) Main input pojde bud cez bateriu alebo ak tam bateria nebude tak sa spoji vnutry už zo senzor board.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

EERCrso, 2021-02-22:
toto sa teda bude zapracovavat alebo nie ?

xosuskyf, 2021-02-22:
Zatial co viem, tak je urobene len uchytenie na novu powerbanku bez zapuzdrenia kablov.