vtothsvk/AP-Home #72: D/N switch possible short circuit

vtothsvk, 2020-11-26:
**Describe the bug**
There's a possibility, the D/N switch shorts VCC to ground for a fraction of a second during switching.

**To do to test**
1. Try to find a position, where all of the switch pins are interconnected
(Proste to treba pípnuť, keď sa to prepína, či to celé neskratuj)

**Expected behavior**
V prípade, že skrauje pri prepínaní, bude sa to dať pípnuť.

mato24101, 2020-11-26:
Zajtra ráno to vyskúšam

EERCrso, 2021-02-22:
problem pretrvava

vtothsvk, 2021-02-23:
@xosuskyf @mato24101 zmerali ste to dakto? Ak nie, tak to zajtra zmeriam.

mato24101, 2021-02-23:
Ja som to meral a nepiplo mi to ale prípadne tam vieme hodiť rezistor a lebo prípadne pripojiť iba gnd na pin a druhý stav by bol floating. Je to riešenie?

xosuskyf, 2021-02-23:
Ak Matove riesenie nie je vhodne, tak navrhujem vyriesit to na urvoni kodu a buduceho manualu, kde vzdy pri spusteni si doska chekne stav D/N a podla toho nastavi tresholdi. Nasledne v manuale pre uzivatela to bude osetrene tak, ze najprv sa nastavi vhodna poloha Day alebo Night a az potom sa zapne senzor box a ak je sensor box uz zapnuty, tak po kazdej zmene D/N switchu je nutne resetovacim tlacitkom resetnut cely sensor box.

vtothsvk, 2021-02-23:
> Ja som to meral a nepiplo mi to ale prípadne tam vieme hodiť rezistor a lebo prípadne pripojiť iba gnd na pin a druhý stav by bol floating. Je to riešenie?

Žial, by to ničomu nepomohlo.

> Ak Matove riesenie nie je vhodne, tak navrhujem vyriesit to na urvoni kodu a buduceho manualu, kde vzdy pri spusteni si doska chekne stav D/N a podla toho nastavi tresholdi. Nasledne v manuale pre [...]

Na úrovni kódu to nemáš ako vyiešiť keď ti to hardresetuje devkit 😆. Ale súhlasím s upozornením na potrebu resetu po zmene režimu.

closed 2021-03-05 by vtothsvk