vtothsvk/AP-Home #74: timer after Mute triggered

EERCrso, 2020-12-02:
**Is your feature request related to a problem? Please describe.**
Aby nedošlo k tomu že mutnem board ale kedže som pred motion tak mi tregerrne error..

**Describe the solution you'd like**
bolo by vhodne vediet nastavit čas počas ktoreho po stlačeni mute buttu budu všetky alerty ignorovane. Toto nastavenie je  potrebne pre night/ day režim

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

vtothsvk, 2020-12-02:
```
/** Alert control directives
 */
#define muteDuration    2000
```
ste si asi nevšimli.

EERCrso, 2020-12-02:
Aha prepač nevšimli... je to v mili sekundach ?  Ale moc to nejde... nastavil som  20 000 spustil alert stlačil gombik .. alert zmizol za chvilu ( ale asi nie koli gombiku ) a hned som mohol trierrnut dalši

vtothsvk, 2020-12-02:
Ale predpokladám, že máte pustený printout, vtedy button nefunguje, lebo mu to breakne časovanie.

vtothsvk, 2020-12-02:
@EERCrso skúsili, ste btw. či hodnotu buttonu číta správne?

vtothsvk, 2020-12-02:
Mne to meria správnu hodnotu na danom pine, bol tam ale malý zádrheľ, a síce button handler bol zle inicializovaný, testol som s káblikom a hralo všetko ako má. Kód najdete na [buttonFix](https://github.com/vtothsvk/AP-Home/tree/buttonFix) (https://github.com/vtothsvk/AP-Home/pull/77)

EERCrso, 2020-12-02:
čo mysliš pod print out ? printout  toto ?  //#define _DEBUG  ?  Nechal som zakomentovane a po stlačeni buttnu nedojde k zmene statusu ani pri jednej polohe switch

ked si zobrazim vypis  a ked stlačim button tam vo vypise mam jednotku...

**@EERCrso skúsili, ste btw. či hodnotu buttonu číta správne?**   Ako to mame skusiť ?

vtothsvk, 2020-12-02:
> **@EERCrso skúsili, ste btw. či hodnotu buttonu číta správne?** Ako to mame skusiť ?
takto
> ked si zobrazim vypis a ked stlačim button tam vo vypise mam jednotku...

Ono inak to trochu pomalšie reaguje treba nechať to tlačítko takú dobrú polsekundu, možno aj sekundu stlačené, kým to zareaguje btw.

EERCrso, 2020-12-02:
Bohužial .. skusil som prave ako vraviš.. a nič stale je alert aj ked dlho aj ked par sekund podržim ... vypis nemam aktivovany.... pozeram sa len na serial monitor či mi nebude vyhadzovať same nuly...

vtothsvk, 2020-12-02:
no keby nemáš aktivovaný výpis tak by to nemalo písať nič, iba, že sa nabootovalo

vtothsvk, 2020-12-02:
@EERCrso mali by ste čas na chvíľu s tým deviceom dobehnúť, že by som sa naň pozrel?

EERCrso, 2020-12-02:
@vtothsvk ideme za tebou

vtothsvk, 2020-12-02:
Tak teda funguje na 2 kliky 😅

closed 2020-12-02 by vtothsvk