vtothsvk/AP-Home #75: No debug monitor update in Night regime

EERCrso, 2020-12-02:
**Describe the bug**
v pripade ak je switch prepnuty na night nedochadza k aktualizacii vypisu.. Alert sa spravaju ako maju  ale vypis je ako  keby zamrnuty  a ukazuje hodnotu pred zmenou switch.. 

**To Reproduce**
Steps to reproduce the behavior e.g.:
1. Flash main
2. swittch to day
3. watch monitor 
4. switch to night
5. see how monitor data  does not change ?

**Expected behavior**
že bude vypisovat hodnoty na senzoroch aj v pripade night režimu

**Evironment:**
 - OS: [e.g. Windows 10]
 - IDE used and its version (e.g. VSCode 1.48.2 with Arduino IDE 1.8.9)
 - Compiler toolchain and version (xtensa-lx106-elf-gcc)
 - https://github.com/vtothsvk/AP-Home/commit/e2b5ffa576947d6ca2d2a40cfbf4d2d7a938a9fc

**Additional context**
Add any other context about the problem here.

vtothsvk, 2020-12-02:
Malo by byť fixnute v [printoutFix](https://github.com/vtothsvk/AP-Home/tree/printoutFix) ([PR #76]
(https://github.com/vtothsvk/AP-Home/pull/76))

EERCrso, 2020-12-02:
nefunguje to.. stale je to to iste

vtothsvk, 2020-12-02:
už musí! 😅 [printoutFix](https://github.com/vtothsvk/AP-Home/tree/printoutFix) (https://github.com/vtothsvk/AP-Home/pull/78)

EERCrso, 2020-12-02:
Funguje zatvaram

closed 2020-12-02 by EERCrso